PkmX/lcamera #123: OnePlus One CyanogenMod 12 (Android 5.0.2) Don't work L Camera!!

AlexanderBlagodatskih, 2015-01-07:
Please add support OnePlus One in this application.

AlexanderBlagodatskih, 2015-01-08:
Logcat

I/Timeline( 9372): Timeline: Activity_launch_request id:pkmx.lcamera time:1582497
I/ActivityManager(  756): START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=pkmx.lcamera/.MainActivity bnds=[540,1275][804,1607] (has extras)} from uid 10023 on display 0
W/ResourcesManager(  756): Asset path '/system/framework/android.test.runner.jar' does not exist or contains no resources.
E/AudioTrack(  756): AudioTrack::set : Exit
D/audio_hw_primary(  231): out_set_parameters: enter: usecase(1: low-latency-playback) kvpairs: routing=2
D/audio_hw_primary(  231): start_output_stream: enter: stream(0xb605f9c0)usecase(1: low-latency-playback) devices(0x2)
D/audio_hw_primary(  231): select_devices: out_snd_device(2: speaker) in_snd_device(0: )
W/msm8974_platform(  231): Codec backend bitwidth 16, samplerate 48000
D/hardware_info(  231): hw_info_append_hw_type : device_name = speaker
I/msm8974_platform(  231): platform_send_audio_calibration: sending audio calibration for snd_device(2) acdb_id(15)
D/ACDB-LOADER(  231): ACDB -> send_audio_cal, acdb_id = 15, path =  0
D/ACDB-LOADER(  231): ACDB -> send_adm_topology
D/ACDB-LOADER(  231): ACDB -> ACDB_CMD_GET_AUDPROC_COMMON_TOPOLOGY_ID
D/ACDB-LOADER(  231): ACDB -> send_asm_topology
D/ACDB-LOADER(  231): ACDB -> ACDB_CMD_GET_AUDPROC_STREAM_TOPOLOGY_ID
D/ACDB-LOADER(  231): ACDB -> send_audtable
D/ACDB-LOADER(  231): ACDB -> ACDB_CMD_GET_AUDPROC_COMMON_TABLE
D/ACDB-LOADER(  231): ACDB -> AUDIO_SET_AUDPROC_CAL
D/ACDB-LOADER(  231): ACDB -> send_audvoltable
D/ACDB-LOADER(  231): ACDB -> ACDB_CMD_GET_AUDPROC_GAIN_DEP_STEP_TABLE
D/        (  231): Failed to fetch the lookup information of the device 0000000F 
E/ACDB-LOADER(  231): Error: ACDB AudProc vol returned = -19
D/ACDB-LOADER(  231): ACDB -> AUDIO_SET_AUDPROC_VOL_CAL
D/ACDB-LOADER(  231): ACDB -> send_afe_cal
D/ACDB-LOADER(  231): ACDB -> ACDB_CMD_GET_AFE_COMMON_TABLE
D/        (  231): Failed to fetch the lookup information of the device 0000000F 
E/ACDB-LOADER(  231): Error: ACDB AFE returned = -19
D/ACDB-LOADER(  231): ACDB -> AUDIO_SET_AFE_CAL
D/audio_hw_primary(  231): select_devices: done
V/ActivityManager(  756): Display changed displayId=0
I/ActivityManager(  756): Config changes=480 {1.0 250mcc1mnc ru_RU ldltr sw360dp w640dp h335dp 480dpi nrml long land finger -keyb/v/h -nav/h s.20}
I/InputReader(  756): Reconfiguring input devices.  changes=0x00000004
I/InputReader(  756): Device reconfigured: id=5, name='synaptics-rmi-ts', size 1080x1920, orientation 1, mode 1, display id 0
I/CameraService(  231): getCameraCharacteristics: Switching to HAL1 shim implementation...
I/CameraService(  231): getCameraCharacteristics: Switching to HAL1 shim implementation...
W/ArrayUtils( 9771): Ignoring invalid value manual
W/ArrayUtils( 9771): Ignoring invalid value manual-cct
W/ArrayUtils( 9771): Ignoring invalid value emboss
W/ArrayUtils( 9771): Ignoring invalid value sketch
W/ArrayUtils( 9771): Ignoring invalid value neon
W/ArrayUtils( 9771): Ignoring invalid value asd
W/ArrayUtils( 9771): Ignoring invalid value backlight
W/ArrayUtils( 9771): Ignoring invalid value flowers
W/ArrayUtils( 9771): Ignoring invalid value AR
V/ActivityManager(  756): Display changed displayId=0
I/InputReader(  756): Reconfiguring input devices.  changes=0x00000004
I/InputReader(  756): Device reconfigured: id=5, name='synaptics-rmi-ts', size 1080x1920, orientation 0, mode 1, display id 0
E/ActivityManager(  756): Invalid thumbnail dimensions: 576x576
I/ActivityManager(  756): Config changes=480 {1.0 250mcc1mnc ru_RU ldltr sw360dp w360dp h615dp 480dpi nrml long port finger -keyb/v/h -nav/h s.21}
I/WindowManager(  756): Screen frozen for +278ms due to Window{3375b024 u0 Starting pkmx.lcamera}
I/MicrophoneInputStream( 9354): mic_close gzi@17740598
D/audio_hw_primary(  231): in_standby: enter: stream (0xb100f980) usecase(7: audio-record)
D/hardware_info(  231): hw_info_append_hw_type : device_name = voice-rec-mic
D/audio_hw_primary(  231): adev_close_input_stream: enter:stream_handle(0xb100f980)
D/audio_hw_primary(  231): in_standby: enter: stream (0xb100f980) usecase(7: audio-record)
I/HotwordRecognitionRnr( 9354): Hotword detection finished
I/HotwordRecognitionRnr( 9354): Stopping hotword detection.
I/art     (  756): Explicit concurrent mark sweep GC freed 54937(2MB) AllocSpace objects, 0(0B) LOS objects, 29% free, 39MB/55MB, paused 1.389ms total 101.419ms
W/InputMethodManagerService(  756): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@121fb29f attribute=null, token = android.os.BinderProxy@279725d8
I/Timeline( 9372): Timeline: Activity_idle id: android.os.BinderProxy@36414e9a time:1583027
I/MicrophoneInputStream( 9354): mic_starting gzi@fab2f0c
D/audio_hw_primary(  231): adev_open_input_stream: enter: sample_rate(16000) channel_mask(0x10) devices(0x80000004)        stream_handle(0xb100f700)
I/HotwordRecognitionRnr( 9354): Starting hotword detection.
I/AudioFlinger(  231): AudioFlinger's thread 0xad416000 ready to run
D/audio_hw_primary(  231): in_standby: enter: stream (0xb100f700) usecase(7: audio-record)
D/audio_hw_primary(  231): in_standby: enter: stream (0xb100f700) usecase(7: audio-record)
D/audio_hw_primary(  231): in_set_parameters: enter: kvpairs=input_source=6;routing=-2147483644
I/MicrophoneInputStream( 9354): mic_started gzi@fab2f0c
[... 413 lines cut ...]
[... cut ...]
V/ActivityManager(  756): Display changed displayId=0
D/SurfaceFlinger(  208): Set power mode=0, type=0 flinger=0xb6a62000
D/qdhwcomposer(  208): hwc_setPowerMode: Setting mode 0 on display: 0
I/qdhwcomposer(  208): handle_blank_event: dpy:0 panel power state: 0
D/qdhwcomposer(  208): hwc_setPowerMode: Done setting mode 0 on display 0
D/SurfaceControl(  756): Excessive delay in setPowerMode(): 242ms
I/QCOM PowerHAL(  756): Got set_interactive hint
E/libsuspend(  756): Error writing to /sys/power/state: Device or resource busy

peshellas, 2015-01-08:
Mate this is a wrong logcat, you provided a logcat of your whole device instead of a logcat only for lcamera, plus you must have not enabled VERBOSE cause no info from what is important like, android.request.availableCapabilities is available on your logcat.

jezuss, 2015-03-09:
yes, can u port to oneplus one. this gona be the best camera ever