PkmX/lcamera #124: Galaxy S4 Official Android 5.0 GPE Support(Log included)

peshellas, 2015-01-08:
Hey mate im giving you the output of the log so you can check if the official android 5.0 for s4 gpe can support the new api. Check and please respond mate. keep up the good work!! :)  I think it still doesnt support it but its better if u check it too.

 1 01-08 15:04:53.577 I/ActivityManager(  788): START u0 {act=android.intent.action.MAIN cat=[android.intent.category .LAUNCHER] flg=0x10200000 cmp=pkmx.lcamera/.MainActivity bnds=[540,960][540,960]} from uid 10068 on display 0   2 01-08 15:04:53.580 I/am_create_activity(  788): [0,707742230,663,pkmx.lcamera/.MainActivity,android.intent.action. MAIN,NULL,NULL,270532608]   3 01-08 15:04:53.721 I/am_restart_activity(  788): [0,707742230,663,pkmx.lcamera/.MainActivity]   4 01-08 15:04:53.762 V/lcamera ( 7678): ===== Camera 0 =====   5 01-08 15:04:53.784 V/lcamera ( 7678): android.colorCorrection.availableAberrationModes   6 01-08 15:04:53.784 V/lcamera ( 7678): 1   7 01-08 15:04:53.784 V/lcamera ( 7678): android.control.aeAvailableAntibandingModes   8 01-08 15:04:53.784 V/lcamera ( 7678): 0   9 01-08 15:04:53.784 V/lcamera ( 7678): 1  10 01-08 15:04:53.784 V/lcamera ( 7678): 2  11 01-08 15:04:53.784 V/lcamera ( 7678): 3  12 01-08 15:04:53.784 V/lcamera ( 7678): android.control.aeAvailableModes  13 01-08 15:04:53.784 V/lcamera ( 7678): 1  14 01-08 15:04:53.785 V/lcamera ( 7678): 2  15 01-08 15:04:53.785 V/lcamera ( 7678): 3  16 01-08 15:04:53.785 V/lcamera ( 7678): android.control.aeAvailableTargetFpsRanges  17 01-08 15:04:53.785 V/lcamera ( 7678): [4000, 30000]  18 01-08 15:04:53.785 V/lcamera ( 7678): android.control.aeCompensationRange  19 01-08 15:04:53.785 V/lcamera ( 7678): [-12, 12]  20 01-08 15:04:53.786 V/lcamera ( 7678): android.control.aeCompensationStep  21 01-08 15:04:53.786 V/lcamera ( 7678): 166667/1000000  22 01-08 15:04:53.786 V/lcamera ( 7678): android.control.afAvailableModes  23 01-08 15:04:53.786 V/lcamera ( 7678): 1  24 01-08 15:04:53.786 V/lcamera ( 7678): 0  25 01-08 15:04:53.786 V/lcamera ( 7678): 2  26 01-08 15:04:53.786 V/lcamera ( 7678): 4  27 01-08 15:04:53.787 V/lcamera ( 7678): 3  28 01-08 15:04:53.788 V/lcamera ( 7678): android.control.availableEffects  29 01-08 15:04:53.788 V/lcamera ( 7678): 0  30 01-08 15:04:53.788 V/lcamera ( 7678): 1  31 01-08 15:04:53.788 V/lcamera ( 7678): 2  32 01-08 15:04:53.788 V/lcamera ( 7678): 4  33 01-08 15:04:53.789 V/lcamera ( 7678): android.control.availableSceneModes  34 01-08 15:04:53.789 V/lcamera ( 7678): 0  35 01-08 15:04:53.789 V/lcamera ( 7678): 2  36 01-08 15:04:53.789 V/lcamera ( 7678): 5  37 01-08 15:04:53.790 V/lcamera ( 7678): 10  38 01-08 15:04:53.790 V/lcamera ( 7678): 14  39 01-08 15:04:53.790 V/lcamera ( 7678): 18  40 01-08 15:04:53.790 V/lcamera ( 7678): 1  41 01-08 15:04:53.790 V/lcamera ( 7678): android.control.availableVideoStabilizationModes  42 01-08 15:04:53.790 V/lcamera ( 7678): 0  43 01-08 15:04:53.790 V/lcamera ( 7678): android.control.awbAvailableModes  44 01-08 15:04:53.790 V/lcamera ( 7678): 1  45 01-08 15:04:53.790 V/lcamera ( 7678): 2  46 01-08 15:04:53.790 V/lcamera ( 7678): 3  47 01-08 15:04:53.790 V/lcamera ( 7678): 5  48 01-08 15:04:53.790 V/lcamera ( 7678): 6  49 01-08 15:04:53.790 V/lcamera ( 7678): android.control.maxRegionsAe  50 01-08 15:04:53.790 V/lcamera ( 7678): 0  51 01-08 15:04:53.790 V/lcamera ( 7678): android.control.maxRegionsAf  52 01-08 15:04:53.791 V/lcamera ( 7678): 1  53 01-08 15:04:53.791 V/lcamera ( 7678): android.control.maxRegionsAwb  54 01-08 15:04:53.791 V/lcamera ( 7678): 0  55 01-08 15:04:53.791 V/lcamera ( 7678): android.flash.info.available  56 01-08 15:04:53.791 V/lcamera ( 7678): true  57 01-08 15:04:53.791 V/lcamera ( 7678): android.info.supportedHardwareLevel  58 01-08 15:04:53.791 V/lcamera ( 7678): 2  59 01-08 15:04:53.791 V/lcamera ( 7678): android.jpeg.availableThumbnailSizes  60 01-08 15:04:53.791 V/lcamera ( 7678): 0x0  61 01-08 15:04:53.791 V/lcamera ( 7678): 176x144  62 01-08 15:04:53.791 V/lcamera ( 7678): 320x240  63 01-08 15:04:53.791 V/lcamera ( 7678): 352x288  64 01-08 15:04:53.791 V/lcamera ( 7678): 432x288  65 01-08 15:04:53.792 V/lcamera ( 7678): 480x288  66 01-08 15:04:53.792 V/lcamera ( 7678): 512x288  67 01-08 15:04:53.792 V/lcamera ( 7678): 512x384  68 01-08 15:04:53.792 V/lcamera ( 7678): android.lens.facing  69 01-08 15:04:53.792 V/lcamera ( 7678): 1  70 01-08 15:04:53.792 V/lcamera ( 7678): android.lens.info.availableFocalLengths

 71 01-08 15:04:53.792 V/lcamera ( 7678): 4.2  72 01-08 15:04:53.792 V/lcamera ( 7678): android.noiseReduction.availableNoiseReductionModes  73 01-08 15:04:53.792 V/lcamera ( 7678): 1  74 01-08 15:04:53.792 V/lcamera ( 7678): android.request.availableCapabilities  75 01-08 15:04:53.792 V/lcamera ( 7678): 0  76 01-08 15:04:53.792 V/lcamera ( 7678): android.request.maxNumOutputProc  77 01-08 15:04:53.792 V/lcamera ( 7678): 3  78 01-08 15:04:53.792 V/lcamera ( 7678): android.request.maxNumOutputProcStalling  79 01-08 15:04:53.793 V/lcamera ( 7678): 1  80 01-08 15:04:53.793 V/lcamera ( 7678): android.request.maxNumOutputRaw  81 01-08 15:04:53.793 V/lcamera ( 7678): 0  82 01-08 15:04:53.793 V/lcamera ( 7678): android.request.partialResultCount  83 01-08 15:04:53.793 V/lcamera ( 7678): 1  84 01-08 15:04:53.793 V/lcamera ( 7678): android.request.pipelineMaxDepth  85 01-08 15:04:53.793 V/lcamera ( 7678): 6  86 01-08 15:04:53.793 V/lcamera ( 7678): android.scaler.availableMaxDigitalZoom  87 01-08 15:04:53.793 V/lcamera ( 7678): 2.0  88 01-08 15:04:53.793 V/lcamera ( 7678): android.scaler.croppingType  89 01-08 15:04:53.793 V/lcamera ( 7678): 0  90 01-08 15:04:53.793 V/lcamera ( 7678): android.scaler.streamConfigurationMap  91 01-08 15:04:53.794 V/lcamera ( 7678): Format 17  92 01-08 15:04:53.794 V/lcamera ( 7678): 1440x1080: 0 0  93 01-08 15:04:53.794 V/lcamera ( 7678): 1280x720: 0 0  94 01-08 15:04:53.794 V/lcamera ( 7678): 1056x864: 0 0  95 01-08 15:04:53.794 V/lcamera ( 7678): 960x720: 0 0  96 01-08 15:04:53.794 V/lcamera ( 7678): 720x480: 0 0  97 01-08 15:04:53.794 V/lcamera ( 7678): 640x480: 0 0  98 01-08 15:04:53.795 V/lcamera ( 7678): 320x240: 0 0  99 01-08 15:04:53.795 V/lcamera ( 7678): 176x144: 0 0 100 01-08 15:04:53.795 V/lcamera ( 7678): Format 256 101 01-08 15:04:53.795 V/lcamera ( 7678): 4128x3096: 0 1107400448 102 01-08 15:04:53.795 V/lcamera ( 7678): 4128x2322: 0 880550336 103 01-08 15:04:53.795 V/lcamera ( 7678): 3264x2448: 0 767309312 104 01-08 15:04:53.795 V/lcamera ( 7678): 3264x1836: 0 625481984 105 01-08 15:04:53.795 V/lcamera ( 7678): 2048x1536: 0 423346688 106 01-08 15:04:53.795 V/lcamera ( 7678): 2048x1152: 0 367510016 107 01-08 15:04:53.796 V/lcamera ( 7678): 1280x720: 0 265433600 108 01-08 15:04:53.796 V/lcamera ( 7678): 640x480: 0 221811200 109 01-08 15:04:53.796 V/lcamera ( 7678): Format 842094169 110 01-08 15:04:53.796 V/lcamera ( 7678): 1440x1080: 0 0 111 01-08 15:04:53.796 V/lcamera ( 7678): 1280x720: 0 0 112 01-08 15:04:53.800 V/lcamera ( 7678): 1056x864: 0 0 113 01-08 15:04:53.800 V/lcamera ( 7678): 960x720: 0 0 114 01-08 15:04:53.801 V/lcamera ( 7678): 720x480: 0 0 115 01-08 15:04:53.801 V/lcamera ( 7678): 640x480: 0 0 116 01-08 15:04:53.801 V/lcamera ( 7678): 320x240: 0 0 117 01-08 15:04:53.801 V/lcamera ( 7678): 176x144: 0 0 118 01-08 15:04:53.801 V/lcamera ( 7678): Format 35 119 01-08 15:04:53.802 V/lcamera ( 7678): 1440x1080: 0 0 120 01-08 15:04:53.802 V/lcamera ( 7678): 1280x720: 0 0 121 01-08 15:04:53.802 V/lcamera ( 7678): 1056x864: 0 0 122 01-08 15:04:53.802 V/lcamera ( 7678): 960x720: 0 0 123 01-08 15:04:53.802 V/lcamera ( 7678): 720x480: 0 0 124 01-08 15:04:53.802 V/lcamera ( 7678): 640x480: 0 0 125 01-08 15:04:53.802 V/lcamera ( 7678): 320x240: 0 0 126 01-08 15:04:53.802 V/lcamera ( 7678): 176x144: 0 0 127 01-08 15:04:53.802 V/lcamera ( 7678): android.sensor.availableTestPatternModes 128 01-08 15:04:53.803 V/lcamera ( 7678): 0 129 01-08 15:04:53.803 V/lcamera ( 7678): android.sensor.info.activeArraySize 130 01-08 15:04:53.803 V/lcamera ( 7678): Rect(0, 0 - 4128, 3096) 131 01-08 15:04:53.803 V/lcamera ( 7678): android.sensor.info.physicalSize 132 01-08 15:04:53.803 V/lcamera ( 7678): 4.5988655x3.40235 133 01-08 15:04:53.803 V/lcamera ( 7678): android.sensor.info.pixelArraySize 134 01-08 15:04:53.803 V/lcamera ( 7678): 4128x3096 135 01-08 15:04:53.803 V/lcamera ( 7678): android.sensor.info.timestampSource 136 01-08 15:04:53.803 V/lcamera ( 7678): 0 137 01-08 15:04:53.803 V/lcamera ( 7678): android.sensor.orientation 138 01-08 15:04:53.803 V/lcamera ( 7678): 90 139 01-08 15:04: [... cut ...] ics.info.maxFaceCount 281 01-08 15:04:53.860 V/lcamera ( 7678): 2 282 01-08 15:04:53.860 V/lcamera ( 7678): android.sync.maxLatency 283 01-08 15:04:53.860 V/lcamera ( 7678): -1 284 01-08 15:04:54.013 I/am_pause_activity(  788): [0,707742230,pkmx.lcamera/.MainActivity] 285 01-08 15:04:54.080 I/WindowManager(  788): Screen frozen for +483ms due to Window{f2cf769 u0 Starting pkmx.lcamera }

286 01-08 15:04:54.197 I/am_destroy_activity(  788): [0,707742230,663,pkmx.lcamera/.MainActivity,finish-idle]

PkmX, 2015-01-08:
Can you format the log? It's kinda hard to read right now.

That said, the values returned here seem to be garbage. For example, it reports `android.info.supportedHardwareLevel` as 58, which is not a valid hardware level. Are you sure there is support for the new api in that release?

peshellas, 2015-01-08:
the log is messed up for sure, i think it doesnt suport the new api, but i think you should take a look at the log to be sure. I will close this issue until i fix the log mate and i will open it again when i got it.

peshellas, 2015-01-08:
Hey mate, i figured out there is some problem that messes up my logcats, and as a workaround now i can save the log to a pdf, so i fixed it and uploaded it to mega, so here it is get it and check mate.

https://mega.co.nz/#!XssRwbBD!tr93L8XLIsBNvJMgJBhKweeFSyILX7YVdoF0-HrJosY

PkmX, 2015-01-15:
The log still shows that the new functionalities aren't supported (`android.info.supportedHardwareLevel == 2 (LEGACY)` on both cameras). Closing this as upstream for now, feel free to re-open it when the situation changes.

closed 2015-01-15 by PkmX