PkmX/lcamera #125: 120fps mode not recording in 120fps

deltabeard, 2015-01-10:
I am using the 120fps system library - I am using the 'L Camera Lib Changer' which seems to be working as expected.

The video resolution '800x600x120 @ 5mbps' is automatically selected by the lcamera app, but even so, the videos recorded are not at 120fps.

[image]

[image]

[image]
The image above shows that the video was recorded at 60 fps. This happened when I turned Auto Exposure off and set it to 1/60.

Thanks, and great work. :)

modbrin, 2015-01-10:
@deltabeard , try using exposure 1/125

deltabeard, 2015-01-10:
The video is recorded at 30fps when changing the exposure to 1/125.

Theliel, 2015-01-12:
You need a good light conditions. 

Set to 1/125 and try outdoor, work fine.