PkmX/lcamera #126: Preview has a cropped image

huusman, 2015-01-13:
Since build v0.3 the preview for both photo and video mode is cropped. The image is only cropped when previewing in the app, I'd say it's cropped 80% on the sensor from the center. When viewing the pictures afterwards the crop is gone and photo's have the same dimensions etc. as the google camera app.

kagwind, 2015-02-22:
Confirm this problem.
The same happened to the google camera, but they finally  changed it.

ZambettiAle, 2015-02-23:
Same problem on nexus 5

Srandista, 2015-02-23:
I did some testing, and only in version 0.1 the preview on display is actually, what you get on photo. In all following version is preview on display different from result captured by camera.

kagwind, 2015-02-24:
Thanks Srandista. I may look into the code based on what you tested.

Srandista, 2015-02-24:
I did comparison through all versions of lcamera, and for reference also for Google Camera. Captured image go first, then comes screenshot of screen in time of shooting. 

Order of pictures: lcamera version 0.1, 0.2, 0.3, dev and Google Camera. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

PkmX, 2015-02-26:
Fixed in the latest dev-build release.

closed 2015-02-26 by PkmX