PkmX/lcamera #130: install problèm

zaknexus, 2015-01-19:
i did all the step exactly like said but nothing happen i cannot able the 60fps and 120fps in L camera option 
, and thx for this awsome work

Theliel, 2015-01-20:
You need N4/N6 in first instance. L Camera work perfectly without any rooted phone, but you need root to get 60/120Hz. You must replace original system/lib file "libmmcamera_imx179.so" with a patched version (60Hz or 120Hz version). Work fine so revise yours steps.

zaknexus, 2015-01-20:
thx for reply but i alredy have root in my nexus 5 with android 5.0.1 and i patch the original libmmcamera_ilx179.so with 120hz that i downloaded i did all the steps bay use root shell, but still cant use 60 or 120 fps in L camera :(

Theliel, 2015-01-20:
have you restarted your phone or killed "mm-qcamera-daemon" and "mediaserver" after replace? are your new libmmcamera_ilx179.so the proper permissions/owner? (root/root 644). Make sure you have replaced with the proper libmmcamera_ilx179.so file too. 

You can try libchanger if you have trouble doing it manually

zaknexus, 2015-01-20:
yes i did all this, restart, kill app, and the system permission fine too, but dont work, mybe i miss somthing imprtant, but later i ll try libchanger
thx a lot for ur help

zaknexus, 2015-01-20:
yes i did all this, restart, kill app, and the system permission fine too, but dont work, mybe i miss somthing imprtant, but later i ll try libchanger
thx a lot for ur help :)