PkmX/lcamera #134: New versions

Slobo2712, 2015-02-05:
I see that latest released verion is on Nov 25, 2014.So there will not be any new verions anymore or i'm missing something?Tnx

Fishywalrus, 2015-02-05:
Also wondering if this project has come to a standstill.. such potential already!

sharathdaniel, 2015-02-06:
I'm also waiting for an update.

Msamii, 2015-02-10:
It didn't stop you guys. Can't you see the most recent commit. It's done 4 days ago. Have patience

vishesh29, 2015-02-11:
Haha. Sorry! Well you've made something we all cherish! It's really very
hard to wait.

PS - Thank You.

Msamii, 2015-02-11:
Haha well I didn't made it. All goes to the awesome man who did it "@PkmX"

vishesh29, 2015-02-11:
I know. Lol. I really wish slow motion for Nexus 6 comes in the upcoming
release.

Msamii, 2015-02-11:
Let's hope the best and congrats you've got Nexus 6 :') I got 5

closed 2015-02-19 by PkmX