PkmX/lcamera #136: UI overhaul feedback

PkmX, 2015-02-19:
I've just commited and uploaded a new dev-build apk in [releases]( https://github.com/PkmX/lcamera/releases/tag/dev-build). This version contains the long-awaited UI overhaul (as well as bulb mode) that has been sitting on my disk for more than a month. As this is an early version so expect things to break very often.

I would like to hear some feedbacks on the UI change. The main problem with the old UI was that it took way too many taps (including the back button which is located at an awkward position in landscape mode) to switch between various setting panels, and it couldn't support portrait modes. This version is designed so that both orientations can be supported simply by rotating the icons.

Some TODOs:

* Support landscape mode by rotating icons/texts
* Need to find out a better way to communicate the functions of each button
* Need to find out a better way to signal the completion of a capture
* Spacings of the controls in the exposure panel

Here is a sample video for those who don't have a L Camera capable device: https://www.youtube.com/watch?v=pBiOYGgt9KE

utkarshns, 2015-02-19:
I couldn't find a way to set exposure to automatic.

PkmX, 2015-02-19:
Tapping either exposure time or ISO text will go back to auto exposure. The reason there isn't an unified AE button is that I'm going to allow shutter speed and ISO to be manually controlled separately. So you can set one as fixed and let the other one vary, like the various priority modes found in most cameras.

Theliel, 2015-02-20:
-yes, setting automatic exposure is not much clear.. 
-grey labels in settings are confusing, seem disabled options, and video resolution use the same (or similar) grey too.
-shooting icon modes are also non intuitive. Video OK, the other three can be ambiguous, specially the bulb mode

on the other hand all features seem to work pretty well, i want test a little more

theotherpossum, 2015-02-21:
An idea for signaling a capture. You know the DSLRs that have a mechanical mirror for the viewfinder that flips up upon taking a picture? How about a short, customizable and optional "Haptic Feedback"?

sharathdaniel, 2015-02-22:
Tried the latest dev build, i can say that the UI is very confusing for starters. Cant even distinguish if the mode is on auto or manual. Also when switching to other modes like bulb, video there is not even a label or toast to notify it.

Msamii, 2015-02-26:
The image is taken with full sensor but it shows in the UI cropped sensor which I think is major bug...

Srandista, 2015-02-26:
Thank you for fixing crop preview issue. Now, only major problem, which I'm having, is that photo is captured only in portrait mode, even if it was shoot in landscape mode.

Everything else for me is quite understandable (after I read readme and some of these comments), only one function was mystery for me and it was bulb mode. This is I think function, which is not recognizable from current icon without explanation.

mfturner, 2015-03-18:
Don't know if you have feedback yet, but I just updated to Android 5.2 on my Nexus 5 as well as your latest dev-build and haven't had a problem.  I like the new UI and the extra features.  The bulb mode was the only icon that I scratched my head about like everyone else, but I can't think of anything better, and my Canon has plenty of head-scratching icons too.  Going in and out of auto vs manual, and the other icons all came naturally to me.  Between DNG, full manual exposure control, burst with AEB and now bulb lcamera has been my camera of choice since late last year, adding MF that works is pretty helpful.  I pretty much quit using open camera, and I mainly use google camera for stitching panoramas, it does that well and gives multiple choices (cylindrical/spherical/rectilinear).  

If I had requests it would be to give the MF control more precision, maybe allow the select of a logarithm function, so that there was more control at long/medium distances, or maybe have a way to change the control bar's sensitivity, or even use a spinning wheel that allowed multiple spins to rack the focus from near to far to add precision.  That, and it would be cool to learn how google camera does panorama stitching - I'd dump them entirely if you could find a way to do that.