PkmX/lcamera #137: HDR video

dadix, 2015-02-20:
Magic Lantern has implemented hdr video which working from what I understanding recording 1 frame with an exposure and the second one with another exposure , than they are combined. Can be implemented this in Lcamera?

A clip is explained here: https://www.youtube.com/watch?v=WvOM3N34DjE

PkmX, 2015-02-20:
In theory, this can be done with the new API (we just request alternating exposure times for each frame). The problem is that this technique effectively halves the frame rate, and without the patched libraries, the camera is limited to 30fps which would become 15fps when halved.

utkarshns, 2015-02-20:
This technique would have the frames even if they are requested as(example):
1. 1/30s and 400iso
2. 1/30s and 1200iso

I guess the output can be a steady 30fps if the exposures time is faster
than 1/30s. The current burst with exposure can manage 30fps if the
exposure time is faster than 1/30s.

utkarshns, 2015-02-20:
Oh yes, my bad.
For HDR video, which would construct one frame from two images, the FPS
can't be more than 15 with out modded libs.