PkmX/lcamera #141: Switing to video mode makes L camera view finder freeze

sharathdaniel, 2015-02-22:
Using the latest dev build, when switching to video mode the view finder just freezes,the buttons work. But when pressing record it just crashes. Using Nexus 5, i don't have any patches file for video capturing.

modbrin, 2015-02-22:
i also have this issue with highest res., 1920x1080 and above seem to work

PkmX, 2015-02-22:
Can you check if this is still a problem with the dev-build I just uploaded? Thanks.

sharathdaniel, 2015-02-22:
Still freezes, then crashes when pressing record.

ervicop, 2015-02-23:
What worked for me was first set the video resolution to 1080p, switch to video mode, then set the video resolution back to 8MP. Viewfinder/recording capabilities were restored.

Srandista, 2015-02-26:
I have after clean installation set 4k recording resolution by default. With this setting, camera freezes after I switch to video recording. After I selected proper resolution for my Nexus 5, all went well.

PkmX, 2015-02-26:
Fixed with @b03f0eb and updated the dev-build apk. This was due to the app defaulting to 3840x2160 resolution (unsupported by N5 obviously) if no previous setting was saved.

closed 2015-02-26 by PkmX