PkmX/lcamera #145: App doesn't work in landscape mode

cotibi, 2015-02-27:
As the title states, the app doesn't go in landscape mode when turning phone (Nexus 5).

jjjuande, 2015-03-02:
I have the same issue. (Nexus 5)

ja-som, 2015-03-03:
Same on my Nexus5 :(
Whole UI of this camera is in portrait mode even if I hold my phone in landscape mode. Tried to take some pictures/videos but they are rotated, if I want to see them normally, I need to rotate them 90° CCW.

cotibi, 2015-03-15:
As of today pictures are ok when taken in landscape mode but camera UI is still stuck in portrait mode upon rotating phone

PkmX, 2015-03-15:
Yes, you gotta wait for me to implement the landscape UI. The orientation of images is a far more critical issue so I pushed this fix first.

cotibi, 2015-03-15:
LOL I was sure you were working on it. I guess I was to fast to comment about it. My compliments for the great app.

meshuggy, 2015-12-12:
I am still not able to record in landscape. Have I missed something?

cotibi, 2015-12-12:
I belive that the developer abandoned the project. It's a shame...