PkmX/lcamera #153: 60/120 FPS

TicTac7x, 2015-03-20:
60/120 FPS mod doesn't work on Nexus 5 with Android 5.1 version.

Theliel, 2015-03-21:
Of course, you need mod the libmmcamera_imx179.so from the new firmware

RobertusIT, 2015-03-22:
Theliel and you know how to edit library or we must wait the update from developer?

Theliel, 2015-03-22:
I don't have time in this moment, but you can try, pkmx explain it:

https://github.com/PkmX/lcamera/tree/master/libmmcamera_imx179_mod

hackthis02, 2015-03-23:
Any chance of getting Nexus 6 support for 120 FPS?

Theliel, 2015-03-25:
I'd modded both version lmy47d thank to PkmX commented code. I only tested 60fps version and work fine. 120fps should work fine too, I will test them tomorrow, and upload/post if working fine.

imx179_patch dont work after all because the new offsets: 

wh32_offset = 0x2324 to 0x2308
conf_offset = 0x2414 to 0x23F8

the rest of the code seen fine, but I didn't used the patch (I directly mod the lib with hex editor), so i can't tell if the patch need some more change.

PkmX, 2015-03-25:
@Theliel Awesome. I will update the binaries if they actually work with the new offsets.

Wicked0ne, 2015-03-25:
I made this a while back: https://github.com/Wicked0ne/LCameraLibChanger If the libs have been updated I will see about adding the news to my app.

Theliel, 2015-03-25:
ok, both work fine. 
120 is less "accurate" respect fps but in the same manner it was before, you need a very good light conditions if you want record up to 120fps (and a 1/125 exposure at least):

http://www39.zippyshare.com/v/7tzNXYP2/file.html

TicTac7x, 2015-03-25:
I can confirm that the 60fps .so is working, but is the video supposed to be "zoomed" in?

utkarshns, 2015-03-26:
Yes, it takes the top left sensor crop. A part of the sensor captures the
image to increase the capture frame rate.

RobertusIT, 2015-03-26:
So also at 60fps, video is zommed? I thought only with 120fps. Anyway i wait the update library to try myself

hackthis02, 2015-03-27:
I'm happy to do testing for the Nexus 6 when/if needed.

Theliel, 2015-03-28:
hackthis02, without any N6 in my hands is not feasible. Reverse engineering can be from easy to a nightmare. I I can't mod files, test, remod again, test again... im sure you can understand it :)

N6 use a similar lib to config camera parameters, in this case:

"libmmcamera_imx214.so"

lib use a similar patterns relative to sensor characteristics too so in theory, and only in theory, applying the same transformations to the lib (that is easy) and modding lcamera code to support n6 60/120+fps, maybe could work fine. 

In the case of N6 (original values extracted from the lib)

Active Pixel: 4.208x3.120
Pixel Array: 5.008x3.190
Sensor Timing: 480.000.000
Output Timing: 465.000.000
fps: 30.0

In first instance, PxmX should add the logic code to lcamera (i doubt lcamera can detect in this moment n6 60/120fps, but this question can be answered by PxmX much better). Secondly, the modded lib is needed. 

If PxmX make a dev/test version with n6 support (logic detection and other stuff) I can send him a hypothetical working lib, but PxmX could have the same problem if he dont have any n6 to test...

hackthis02, 2015-04-03:
I completely understand. It's just one of the options I miss from my DROID MAXX. If you guys need a test subject I'm happily offering my services.

humanbagel, 2015-04-09:
Does LMY47D use the exact same libs as well?

Theliel, 2015-04-09:
last version is LMY47I, but use the same file that LMY47D

humanbagel, 2015-04-09:
I'm just curious, how come the quality of the 120fos mod isn't very clear?

utkarshns, 2015-04-09:
Because it takes a top left crop of the sensor measuring 800x600.
You will find it grainy and horribly zoomed.
Grainy because the exp. time is faster than 1/125s; which means very less
light and *high ISO*.
Zoomed in because of the cropping.

Solutions:

1) The centre of the lens is the best produce sharp images. Corners suffer
from optical problems. So, instead of the top left crop if the center could
be used, the quality will improve.
2) If pixel-binning could be implemented, the quality would improve greatly.

Both 1 and 2 are probably technically difficult, otherwise they would have
been implemented by now.

PS: The iPhone 6 and 6 Plus use pixel binning to churn out 240fps clips.

vishesh29, 2015-04-13:
When will the Nexus 6 have slow motion feature?

hackthis02, 2015-04-21:
Any chance of the Moto X pro system img helping with this? I believe it has slowmo just like the Moto X 2014.

humanbagel, 2015-05-01:
So, I tried toying around a bit with the LCamera video, currently still running 5.1 LMY47D and whenever I pan over and just scan around the scene, the video has a tendency to glitch out randomly. Is it because of the bug that's a part of OS 5.1?

krock5230, 2015-05-15:
I have installed the lrx21o modded lib over android 5.0.2 lrx22g but doesn't work. The camera work normaly but not allow me to select the x60 fps, and i don't see the 120fps on the app

Theliel, 2015-05-21:
My posted mod work fine for lmy48b. lmy48b use again the same lib, bit-to-bit

MessedupBirdie, 2015-10-17:
Nexus 6, Any help with slowmo and 60fps 1080p?

hackthis02, 2015-11-02:
I finally build the program but I'm getting wrong data

C:\Users\tmassing\Desktop>libmmcamera.exe libmmcamera_imx214.so
libmmcamera_imx214.so: active_pixel_width=-8583 active_pixel_height=-1 pixel_array_width=11836
pixel_array_height=0 sensor_timing=-8586 output_timing=11844 fps=-nan

@Theliel  can you send me copy of the correctly build program?

PkmX, 2015-11-03:
@hackthis02 I noticed that you are using the library for IMX217, but this patch is written specifically for the IMX179 library on Android Marshmallow (MRA58K), so it will not work.

Also, the patcher has been updated to Android 6.0 a few weeks ago, so I guess this issue can be closed. Please open a new issue when a new version of the library is released.

closed 2015-11-03 by PkmX

Prajithp, 2016-07-13:
does anybody have compiled version of imx179_patch? am not able to compile this on mint 17.