PkmX/lcamera #157: messing up with system files

karthi1, 2015-03-29:
i m messing up with system files here.
the installation gone wrong somewhere.
i dont know why camera isnt starting . can i have a detailed installation procedure or a video tutorial ???

Theliel, 2015-03-30:
Installation?? lCamera is a apk application, lcamera dont touch system files. If you have overwrite or deleted system/lib file, restore the original file and permissions (for the same file).

karthi1, 2015-03-30:
Then how to deal with the 60/ 120fps  mod ..? Any detailed steps ?

Theliel, 2015-03-30:
have you read PkmX instructions?? He details step by step in home how you need replace the file. 

Check first if libmmcamera_imx179.so is on place and check again for permissions (use any file manager), should be user: root group:root and 644.

options:

a) File is missing on /system/lib -> you must copy ori file here, use any filemanager and set permissions
b) File is pressent -> make sure permissions are correct
c) File is present & permissions are fine -> Maybe you are using L5.1 and file posted on home are only for 5.0.x

karthi1, 2015-03-30:
I did replacing the original file , set permission , rebooted ...
I m using 5.0.1 n rooted ...
Tried these steps again ... I m not even able to use original library..I m gonna try after  flashing n root again