PkmX/lcamera #174: Android 6 (Nexus 5x/6p) support

Fishywalrus, 2015-10-01:
Hi PkmX
Thanks so much for all the effort you've put into Lcamera.. It's allowed me to make amazing images with my Nexus6!
Are there plans for support for the new Nexus phones when they hit the market?
Regards

iRub1Out, 2015-10-01:
There hasn't even been development on the N6, why would there be any work
done on the newer ones?

Many of us are willing to help but its a dead build it seems.

MessedupBirdie, 2015-10-17:
Hey, Its a really nice app, but i think the girls out here need to put a little more "Barbie stuff" in there. 
ps, i'm writing this comment after using it on my Nexus 6 and i gotta tell ya, its dead slow.

PkmX, 2015-10-19:
It should work with the new nexus (or any other devices that support the camera2 API properly) without any modification, though I'd be interested to see a [dump](https://github.com/PkmX/lcamera#does-it-run-on-nexus-4710-or-other-phones-that-have-received-lollipop-update) of the capabilities of both nexuses to see if they support any new interesting features.

Hexxeh, 2015-10-21:
@PkmX here you go: http://pastebin.com/iFtb17JH

parkerlreed, 2015-10-23:
@Hexxeh :D Long time no see. Which device did you run that for?

Hexxeh, 2015-10-23:
@parkerlreed that's for a Nexus 5X, international version.

parkerlreed, 2015-10-23:
If it makes a difference (probably not) here's the 5X US http://pastebin.com/WDvyKY9J

Fishywalrus, 2015-10-23:
To those with a 5x or 6p, have you been able to successfully use Lcamera on them?

parkerlreed, 2015-10-24:
Yeah it launches and I was able to capture a dng. Didn't try fine tuning.

Fishywalrus, 2015-10-24:
Ah cool.. Is it able for example, to capture a burst set of like 7 shots bracketed? If so, did it take like 10seconds for the files to be written to storage?
Cheers

Fishywalrus, 2015-11-04:
Tried using a 6p with lcamera. 
AF doesn't work but MF does.
Minimum ISO is 80 (not 40 like the nexus 6)
DNG write times are far worse than the n6. The 6 can write 10x 25MB DNGs in 24s, 6p writes 10x 23.6MB DNGs in 53s. It doesn't make sense why write times would be worse on faster and newer NAND flash memory.
The live view area on screen is much larger (better) with the 6p than the 6.

dbau, 2015-11-18:
Hi, 
It works fine on the nexus 5x, but the images are stored upside down. I guess it depends on the position of the sensor? 
Could this be fixed? 

Great app and thanks for the effort!