PkmX/lcamera #175: No more updates?

flopponator, 2015-10-27:
Hey PkmX,

are you still developing this app or have you abandoned this?

balzathor, 2015-10-27:
[As he said](https://github.com/PkmX/lcamera/issues/174#issuecomment-149285756)
It should work with any device having the new camera API, else for the [Nexus 5 60/120FPS](https://github.com/PkmX/lcamera#60120-fps-recording) recording as that requires root and is only limited to Nexus 5. If he has to update it, he only needs to update some standard libs.

Regards.

flopponator, 2015-10-27:
I know but i mean actively developing New functions and fixing bugs (tiny viewfinder on the nexus 6)

balzathor, 2015-10-27:
Yes, if you provide the dev [camera dumps](https://github.com/PkmX/lcamera#does-it-run-on-nexus-4710-or-other-phones-that-have-received-lollipop-update) from the system maybe he will fix it.