PkmX/lcamera #178: Compatibility with Samsung Galaxy S6 Edge+

citytrader, 2015-11-04:
I will be glad if you can make this camera compatible with Samsung Galaxy S6. here is the dump:

Camera module HAL API version: 0x100

Camera module API version: 0x203

Camera module name: Samsung CameraHardware Module

Camera module author: Samsung

Number of camera devices: 2

Vendor tag definitions:

    Dumping configured vendor tag descriptors: 8 entries

      0x80000000 (liveHdrLevelRange) with type 1 (int32) defined in section samsung.android.control

      0x80000001 (liveHdrLevel) with type 1 (int32) defined in section samsung.android.control

      0x80000002 (meteringAvailableMode) with type 1 (int32) defined in section samsung.android.control

      0x80000003 (meteringMode) with type 1 (int32) defined in section samsung.android.control

      0x80000004 (pafAvailableMode) with type 0 (byte) defined in section samsung.android.control

      0x80000005 (pafMode) with type 1 (int32) defined in section samsung.android.control

      0x80010000 (opticalStabilizationOperationMode) with type 1 (int32) defined in section samsung.android.lens

      0x80020000 (availableOpticalStabilizationOperationMode) with type 1 (int32) defined in section samsung.android.lens.info

Camera 0 static information:

  Facing: BACK

  Orientation: 90

  Device version: 0x302

  Device static metadata:

    Dumping camera metadata array: 84 / 126 entries, 4776 / 8416 bytes of extra data.

      Version: 1, Flags: 00000000

      android.colorCorrection.availableAberrationModes (00004): byte[1]

        [0 ]

      android.control.aeAvailableAntibandingModes (10012): byte[2]

        [0 3 ]

      android.control.aeAvailableModes (10013): byte[4]

        [0 1 2 3 ]

      android.control.aeAvailableTargetFpsRanges (10014): int32[10]

        [15 15 24 24 ]

        [10 30 15 30 ]

        [30 30 ]

      android.control.aeCompensationRange (10015): int32[2]

        [-4 4 ]

[... 1638 lines cut ...]
      android.info.supportedHardwareLevel (150000): byte[1]

        [LIMITED ]

      android.sync.maxLatency (170001): int32[1]

        [PER_FRAME_CONTROL ]

  Device is closed, no client instance

Extra Device Information:

  00000

  FFFFF

No active camera clients yet.

---- CameraService Activity Log ----

Current time: [2015:11:04 07:34:21], 48646648

Camera traces (0):

  No camera traces collected.

PkmX, 2015-11-04:
Can you paste the output on something like pastebin or GitHub's gist and post the link back here? The unformatted output makes it quite hard to follow.

citytrader, 2015-11-04:
I edited the post with pastbin link, let me know if its ok!

citytrader, 2015-11-04:
I'm a pro photographer so if you need to ask more questions, your are welcome!